IPCConnectedFactoryExchange/CFX #8: Broadcast Request handling

alexisfouquet, 2018-07-09:
Hi,

At this point, only one message is concerned by Broadcast Request (WhoIsThereRequest).
Is the current CFX code able to handle it?

The request function is defined like this :
`public CFXEnvelope ExecuteRequest(string targetUri, CFXEnvelope request);`

So it allows to only get one answer of one answering end point, not several.

Thanks for your feedback,

Alexis FOUQUET
R&D Software Engineer, EUROPLACER.

engrx2019, 2019-05-17:
Hello,

I am getting an error {"Attach rejected: {address_not_utf8_string,undefined}"} with the following.

Dim s1 As String = String.Format("amqp://{0}", "localhost")
result = theEndpoint.ExecuteRequest(s1, request)

What can I do to resolve this issue?

Thanks

simon-smith, 2019-05-20:
If you replace "localhost" with "127.0.0.1" do you still get the error?

engrx2019, 2019-05-20:
Yes, got the same error. Please see the attached file for error description.
[Doc1.docx](https://github.com/IPCConnectedFactoryExchange/CFX/files/3198576/Doc1.docx)

simon-smith, 2019-05-20:
Have you tried downloading and the APEX CFX Demo? This should give to a good start point.

The code below is how I create my initial connection.
```
        string myHandle = "MyCompany.MyMachineModel.MyMachineSerialNumber";
        string myBroker = "amqp://cfx.aiscorp.com:5672";
        string myExchange = "/exchange/AegisCloud";
            theEndpoint = new AmqpCFXEndpoint();
            theEndpoint.Open(myHandle);
            theEndpoint.AddPublishChannel(new Uri(myBroker), myExchange);
```
The other thing to check is the construction of your JSON message. The message should look something like this:
```
{
   "Version":"1.0",
   "TimeStamp":"2019-05-20T16:38:41.806+00:00",
   "UniqueID":"5212A10E-9E91-4B57-8448-BB00C3ECB667",
   "Source":"pillarhouse.synchrodex.sd278",
   "Target":null,
   "RequestId":null,
   "MessageName":"CFX.GetEndpointInformation",
   "MessageBody":{
      "CFXHandle":"Pillarhouse.Fusion.PX999"
   }
}
```
Hopefully this helps you get a little further.

Simon