IPCConnectedFactoryExchange/CFX #23: Where is ResourceState.On / .Off defined?

PhotonDan, 2019-03-29:
I am attempting to implement the example code in LabVIEW (so apologies for the lack of .NET fluency), and when I look at sending a StationStateChanged message as shown on this [page](http://www.connectedfactoryexchange.com/CFXDemo/sdk/html/cfb57aac-b696-4c6c-b94b-f034b37daf26.htm), I see NewState = ResourceState.On, OldState = ResourceState.Off.

In looking at the [source for ResourceState](https://github.com/IPCConnectedFactoryExchange/CFX/blob/master/CFX/Structures/ResourceState.cs) I do not see On and Off defined.

Are these just assumed Boolean constants (On = 1 and Off = 0)?

Would it be poor practice to send a message about a Station State and not use one of the defined ResourceState enumerations?

Thanks!

simon-smith, 2019-04-01:
Hi
I am not sure you fully understand the methodology for the StationStateChanged and StageStateChanged messages.
A typical message would be:
{
  "OldState": 2200,
  "OldStateDuration": "00:01:25",
  "NewState": 1100,
  "RelatedFault": null
}
Where the OldState and NewState enumerations are defined here. [http://www.connectedfactoryexchange.com/CFXDemo/sdk/html/T_CFX_Structures_ResourceState.htm]
So in the example above:
OldState was "Standby. No Product (Resource is Starved)"
New State is "Productive Time. Regular Work"
It is not a case of state On or Off, but what condition the system of stage is currently in and what was it's previous state.

I hope this helps.
Simon

PhotonDan, 2019-04-01:
Thanks for the response!

I do understand that ResourceState is an enumeration of values corresponding to system condition descriptions, but what I was stuck on is that within the list of enumerated values you have things like SBY_NoProduct = 2200 and  PRD_RegularWork = 1100, but there is no value defined for On or Off... yet the On and Off appear in the example code shown on the Getting Started with the SDK page.

private void btnSend_Click(object sender, EventArgs e)
{
     CFXEnvelope msg = new CFXEnvelope(new StationStateChanged()
     {
          **NewState = ResourceState.On,
          OldState = ResourceState.Off,**
          OldStateDuration = TimeSpan.FromMinutes(43)
     });

     theEndpoint.Publish(env);
}

I was trying to determine if it is considered acceptable practice to have ResourceState values that are not explicitly listed on the original enumeration definition.

simon-smith, 2019-04-01:
Hi 
I now realise where you got the example code from. This is example is pre-CFX agreement. It is from the early days of when we were looking at message structures. I have asked that this example be removed from the documentation.
Currently there is no mechanisium for adding alternative State enumerations. Having said that we are looking to add new states in V1.1. I already have a list of additional states required. If you let us know what you would like adding the A-team (the CFX overwatch team) will look at them and add as required.
Sorry for the confusion. 
Simon