IPCConnectedFactoryExchange/CFX #31: OverallResult PCBSelectiveSoldered

oezguer-cirakoglu, 2019-07-24:
Hello,

I have seen that for example UnitsProcessed has a property "OverallResult", why PCBSelectiveSoldered does not have this property?

BR
Özgür

ChrisJorgensen, 2019-07-24:
Ozgur,

Our CFX team has received this message and is looking into it for a response. We will get back to you shortly.

Chris

simon-smith, 2019-07-25:
Ozgur
Have you implemented the PCBSelectiveSoldered message yet in your system? This was one of the very early messages from before the current structure was created. Selective Soldering results should really be sent under UnitsProcessed as you have pointed out.
The CFX A-Team are looking at this and may move this data the UnitsProcessed message as you suggest for the next release.

oezguer-cirakoglu, 2019-07-26:
Thanks for your Feedback.

Yes, I have implemented the PCBSelektiveSoldered message, Michael Ford suggested this message.

I have checked the "UnitsProcessed" object. there is an an overallresult field, BUT the "ProcessData" for "CommomProcessData" and "UnitProcessData"object is completely empty.

simon-smith, 2019-07-26:
Hi.
Following a conference call with the CFX A-team yesterday we are looking at adding a message structure for Selective Soldering for use with UnitsProcessed. If you look at the examples for UnitsProcessed you will see there are  structures for ReflowProcessData, CoatingProcessData and ReflowProfilingProcessData. We are looking at adding Selective, Wave, Laser, Solder Paste and possibly other processed.
We have a series of conference calls over the next 4-6 weeks when we hope to resolve this. Whatever the desicion is the original message will still exist to maintain backwards compatability.

oezguer-cirakoglu, 2020-03-31:
Hello,

dou you have any updates on this topic?

Which objects should be used for
- Selekctive soldering
- Reflow soldering
- Wave soldering
- ...