IPCConnectedFactoryExchange/CFX #32: Having a machine in CFX Demo List

syrandel, 2019-07-24:
I have already submitted a question about testing the online CFX Demo broker.
What should we do, at ASTER Technologies, to have a machine on the list ?
How can I test consuming messages from the online Aegis broker ?

No one answered. I addressed the message to Michael Ford, and got no answer.
At least tell me to whom should I address my request.

Chris, can you help ?

Thank you

simon-smith, 2019-07-24:
Hi. I have already forwarded you original message to the Aegis guys that can resolve this. We have an online call this week aand I will bring this up for you.
Sorry you haven't had a response prior to this.
Simon

syrandel, 2019-07-24:
Thanks a lot.
Daniel

closed 2019-10-21 by syrandel