IPCConnectedFactoryExchange/CFX #39: Database schema for CFX

syrandel, 2019-09-24:
Hi,

I am considering storing CFX Data in a database; Has someone done it already ?
I am interested mostly in the TestAndInspection class.
If it is the case, can he share a create script for it ?

Thank you

simon-smith, 2019-09-24:
Hi Daniel
I have a simple .NET program that gets messages from a RabbitMQ message broker and stores in a MS SQL Database. It is a bit quick and dirty, but I'm happy to share if that is of help.
Simon
s.smith@pillarhouse.co.uk

syrandel, 2019-09-24:
Hi Simon,
I began to write a DB schema for CFX and realized quickly it is a big task.
I will contact you by email.
Thank you
Daniel

syrandel, 2019-09-24:
Hi,
I just want to clarify. What I want to develop is a sql create script, generating a database including the tables corresponding to all the CFX classes. For example, We will have a table called InspectedUnits having as fields: UnitIdentifier, UnitPositionNumbern OverallResult, ...
It seems to me it is a must for anyone consuming CFX messages and intending to use the data for reports, traceability, ERP systems, diagnostic and repair software tools, ...

syrandel, 2019-10-21:
Nobody answer this one. However, a software consuming CFX messages and processing objects with such fields as TransactionId, WorkOrderId, UniqueIdentifier, will have a hard time finding what refers to what without a database. It is difficult to imagine the CFX developers don't have such a schema.