IPCConnectedFactoryExchange/CFX #46: CFX Unable to consume messages from RabbitMQ queue.

xXSnowBunny, 2019-10-22:
Hi,

I am new in CFX and RabbitMQ.
I had followed the tutorial on both the CFX and also the RabbitMQ to publish and consume messages.

My broker is setup as follows.
1) I had manually created my exchange, queue and bindings in RabbitMQ Broker = 
     exchangeA -> routingKeyA -> queueA (Durable queue , without any additional arguments parameters included)

2) CFX =  
       Uri uri = new Uri($"amqp://{Username}:{Password}@localhost:5672");
            string exchangeAddress = $"/exchange/exhangeA/routingKeyA";
            string queueAddress = $"/queue/queueA/";

     endpoint.AddPublishChannel(uri,exchangeAddress);
     endpoint.AddSubscribeChannel(uri,queueAddress);

Result: 
1) I am able to publish and receive message using CFX successfully 

However if I delete the queue and recreate it using the rabbitMQ management, and make it into a lazy queue by putting ärguments with "x-queue-mode" = true, my CFX is able to publish messages , but it is unable to consume messages from the queue. 

I had already set up the bindings for the queue and exchanges.

I would appreciate it if someone will kindly please guide me / let me know if there is a way to consume message from the queue , if the queue created in the RabbitMQ broker consist of arguments? Or currently CFX is unable to consume message from the queue if the queue was created with arguments, and it is only able to consume message from the queue which is durable without any arguments set?

Kindly please correct me if I am wrong. 

Any feedback would be appreciated. 

Thanking you in anticipation.
Regards,
Snow Bunny.

syrandel, 2019-10-22:
Hi,

 

Can you explain why do you need a lazy queue ?

I did not try CFX methods on a queue with arguments yet; I’ll try it when I have time.

 

Daniel

xXSnowBunny, 2019-10-22:
Hi Daniel,

Thank you for your reply.

According to RabbitMQ:

> Whenever the broker considers it needs to free up memory, messages from this cache will be paged out to disk. Paging a batch of messages to disk takes time and blocks the queue process, making it unable to receive new messages while it's paging. 
> 
> Lazy queues attempt to move messages to disk as early as practically possible. This means significantly fewer messages are kept in RAM in the majority of cases under normal operation. This comes at a cost of increased disk I/O.

https://www.rabbitmq.com/lazy-queues.html 

By enabling LazyQueue , there will not be a sudden change in performance drop.

syrandel, 2019-10-22:
Hi,

 

And do you think the rate and volume of CFX messages published by an assembly line of machines justify the need for such a mechanism ?

simon-smith, 2019-10-22:
If I remember correctly a RabbitMQ broker has been used for each of trade show demo systems and also the test system that Aegis has kindly provided. I am not aware of any slowdown issues. If lines were generating 1000’s/sec then this could be an issue. But if a line was producing this many messages then you could either clustered brokers or look at a commercially hosted solution.

Regards

Simon Smith

Pillarhouse International Ltd.

This email and any files transmitted with it are confidential and intended solely for the use of the individual or entity to whom they are addressed. If you have received this email in error please notify us and remove it from your system. Finally, the recipient should check this email and any attachments for the presence of viruses. Pillarhouse International Ltd. accepts no liability for any damage caused by any virus transmitted by this email.

Pillarhouse International Ltd. is a limited company, registered in England and Wales.
Registered Office: 32 Queen Anne Street, London W1G 8HD, England. Registered No.: 1201305

xXSnowBunny, 2019-10-22:
Hi Daniel & Simon,

Thank you for your reply.

For now I am just testing the CFX and trying to consume the message from a queue that contains arguments. It do not necessarily need to be a lazy queue, but a queue that contains other arguments.  I am considering the lazy queue, in case there are too many messages in which will cause RabbitMQ to start flushing the messages into the disk. 

Thank you .

Abhishek150995, 2020-02-28:
theEndpoint.AddSubscribeChannel(new Uri(myBroker), queueAddress); adds a Subscriber to the AMQP. 
Which method can be used to receive the message??

xXSnowBunny, 2020-03-04:
Hi @Abhishek150995 
Thank you for the reply. 

Yup , I can subscribe to the queue using the code but that is only if the queue is a durable queue without any arguments set the queue. 

And if I were to use the same code to subscribe to a queue that have arguments, lets say the queue had been pre-created in the RabbitMQ broker with "auto delete = true", then it will not be able to subscribe to the queue.

An error as shown below, will be thrown:

> 
"ERROR extablishing connection to queue = PRECONDITION_FAILED - inequivalent arg 'auto_delete' for queue 'queueA' in vhost '/': received 'false' but current is 'true'"

Abhishek150995, 2020-03-07:
> Hi @Abhishek150995
> Thank you for the reply.
> 
> Yup , I can subscribe to the queue using the code but that is only if the queue is a durable queue without any arguments set to the queue.
> 
> And if I were to use the same code to subscribe to a queue that have arguments, lets say the queue had been pre-created in the RabbitMQ broker with "auto delete = true", then it will not be able to subscribe to the queue.
> 
> An error as shown below, will be thrown:
> 
> > 
> 
> "ERROR establishing connection to queue = PRECONDITION_FAILED - inequivalent arg 'auto_delete' for queue 'queueA' in vhost '/': received 'false' but current is 'true'"
> 
> I am wondering if CFX have any parameter where I can use to set the queue arguments as in RabbitMQ Client library which allows me to set the queue arguments before it tries to subscribe to the queue. Its because the queue arguments from CFX Client do not match to the Queue arguments that had been pre-created in the RabbitMQ, so when it tries to establish a connection and subscribe to the queue, it will throw an error.

Thanks for your valuable reply.

Can you share a snippet of your subscriber code? 
I am unable to subscribe to the published messages.
AddSubscriber function returns IsOpen=false.
Also can you please explain to me the use if AddListner function?

Thanks again.

StevenAZy, 2021-09-01:
> Hi,
> 
> I am new in CFX and RabbitMQ.
> I had followed the tutorial on both the CFX and also the RabbitMQ to publish and consume messages.
> 
> My broker is setup as follows.
> 
> 1. I had manually created my exchange, queue and bindings in RabbitMQ Broker =
>    exchangeA -> routingKeyA -> queueA (Durable queue , without any additional arguments parameters included)
> 2. CFX =
>    Uri uri = new Uri($"amqp://{Username}:{Password}@localhost:5672");
>    string exchangeAddress = $"/exchange/exhangeA/routingKeyA";
>    string queueAddress = $"/queue/queueA/";
>    endpoint.AddPublishChannel(uri,exchangeAddress);
>    endpoint.AddSubscribeChannel(uri,queueAddress);
> 
> Result:
> 
> 1. I am able to publish and receive message using CFX successfully
> 
> However if I delete the queue and recreate it using the rabbitMQ management, and make it into a lazy queue by putting arguments with "x-queue-mode" = true, or enable auto-delete to true for the queue, my CFX is able to publish messages , but it is unable to consume messages from the queue.
> An error message will keep showing that as follows:
> 
> "ERROR extablishing connection to queue = PRECONDITION_FAILED - inequivalent arg 'auto_delete' for queue 'queueA' in vhost '/': received 'false' but current is 'true'"
> 
> I had already set up the bindings for the queue and exchanges.
> 
> I would appreciate it if someone will kindly please guide me / let me know if there is a way to consume message from the queue , if the queue created in the RabbitMQ broker consist of arguments? Or currently CFX is unable to consume message from the queue if the queue was created with arguments, and it is only able to consume message from the queue which is durable without any arguments set?
> 
> Kindly please correct me if I am wrong.
> 
> Any feedback would be appreciated.
> 
> Thanking you in anticipation.
> Regards,
> Snow Bunny.

Hi, @xXSnowBunny
Thank you for your contribution!
I am also new in CFX and RabbitMQ, so I want to know how do you created my exchange, queue, and bindings in RabbitMQ manually?
Best Regards!
Stevenazy