IPCConnectedFactoryExchange/CFX #47: CFX for Python

xXSnowBunny, 2019-11-06:
Hi,

May I know if there are any plugins for CFX in python as in C# and also sample codes on establishing the connection to the broker? 

The documents in this link is are all only for C#.
http://www.connectedfactoryexchange.com/CFXDemo/sdk/html/cfb57aac-b696-4c6c-b94b-f034b37daf26.htm

So I had been wondering if there are any plugins that I can use for python.

Thank you .

Regards,
Snow Bunny.

ChrisJorgensen, 2019-11-06:
We have a group within the CFX Subcommittee that has been working on an SDK for Python. This group is open to anyone to participate, so let me know if you would be interested in joining. Otherwise, once that SDK is available, we will post it to Github.

Chris

http://www.connectedfactoryexchange.com/CFXDemo/sdk/html/cfb57aac-b696-4c6c-b94b-f034b37daf26.htm

xXSnowBunny, 2019-11-07:
Hi @ChrisJorgensen ,

Thank you for your reply and also for the information. 

I will be looking forward to the Python SDK release =) .

Thank you.

Regards,
SnowBunny.

f4b0, 2022-04-04:
Hi @ChrisJorgensen,
 whats the current state of the python integration? 
Is this still being worked on?

Thanks
Best Regards
Fabian

xiaolingsi, 2025-11-17:
> Hi [@ChrisJorgensen](https://github.com/ChrisJorgensen) ,
> 
> Thank you for your reply and also for the information.
> 
> I will be looking forward to the Python SDK release =) .
> 
> Thank you.
> 
> Regards, SnowBunny.

Hello，look here
 https://github.com/xiaolingsi/CFX