IPCConnectedFactoryExchange/CFX #54: Recommendation to Set Application Headers on AMQP 1.0 Messages

timburke, 2020-02-20:
## Overview

Currently, the AMQP 1.0 serialization of CFX messages appears to consist of:

1. Serializing the message to `json` [source](https://github.com/IPCConnectedFactoryExchange/CFX/blob/master/CFX/CFXEnvelope.cs#L158)
2. Inserting the message into a `json` envelope that contains the message topic, source, target, id and a few other fields.  [source](https://github.com/IPCConnectedFactoryExchange/CFX/blob/master/CFX/CFXEnvelope.cs#L16)
3. Sending the envelope as an opaque message through the AMQP 1.0 protocol.  [source](https://github.com/IPCConnectedFactoryExchange/CFX/blob/master/CFX/Transport/AmqpUtilities.cs#L17)

This means that there is no information about the contents of the message available in a format that can be used to route or filter the messages based on, for example, topic.

Each machine can be configured to send all messages generated to a single fixed `target` address but, if that target is a message broker, there is no exposed information to allow the broker to route individual messages to different queues.  An example use case would be:
> There are two systems interested in different CFX messages.  One system just wants placement data to show a realtime line dashboard.  It is a small computer, perhaps a low-cost IoT linux device that expects to receive a low volume of CFX messages.
>
>  There is a second system that performs level 4 traceability and receives detailed reports of all components installed on each production unit.

As I understand it, there is no way currently to set up a single amqp 1.0 broker that would be able to receive all messages from a machine and route the performance topic only to system 1 and all messages to system 2.  

As the size of CFX installations grow, the implicit use of fanout-only exchanges in this way could cause rapid resource need expansion or complex manually configuration of machines in order to have multiple targets with specific messages sent to each target.  This latter setup partially negates the benefit of using a pub/sub compatible protocol like amqp 1.0 which inherently has a way to achieve this goal without needing multiple links to each machine.

The current issue blocking such a centralized cfx broker deployment is that the cfx message to amqp 1.0 mapping does not appear to include metadata like topic and cfx version in a way that the amqp broker can access to make routing decisiosn.

## Potential Solution

> This a potential solution to the above issue proposed here for discussion and brainstorming with the CFX team.

Include key information from the cfx envelope in the `amqp 1.0` "application properties" section of the message.  See [amqp 1.0](http://www.amqp.org/sites/amqp.org/files/amqp.pdf) Page 80.

This header section is reserved by amqp 1.0 for applications like cfx to include a key/value map of information that is designed to be used to route messages without parsing the message contents.

If, for example, the cfx message topic, source, target and cfx version were duplicated as application properties, then you would be able to route a subset of messages based on topic to a subset of consumers.

### Benefits for on-premise RabbitMQ Deployments

RabbitMQ has a mapping of amqp 1.0 application headers to amqp 0.9.1 `headers` which allows them to be used with a rabbitmq header exchange to perform complex filtering and routing.  By adding the cfx topic to application headers then users would be able to set up complex routing scenarios needed by large deployments with no backwards-compatibility implications. 

See [here](https://github.com/rabbitmq/rabbitmq-amqp1.0#message-properties-annotations-headers-etc)

### Benefits to Cloud Deployments

I believe that cloud amqp 1.0 compliant messaging systems also look at application headers to allow for programmatic routing.  For example, Azure Service Bus maps `application headers` to User Properties (see [here](https://docs.microsoft.com/en-us/azure/service-bus-messaging/service-bus-messages-payloads)).  

User Properties can be used to filter message subscriptions.  (see [here](https://docs.microsoft.com/en-us/azure/service-bus-messaging/topic-filters)).

## Bottom Line

Adding key information to the application properties section of the AMQP messages (in addition to placing it in the `json` envelope wrapper) will facilitate large CFX deployments since it permits brokers to route subsets of messages.  

This benefit comes at no cost to point-to-point installations (or small fanout style exchanges) that do not need to worry about resource optimization.

Adding this information sooner rather than later can hopefully ensure that the majority of CFX implementations adopt it so that their messages can be routed in a manner other than point-to-point or fanout.

syrandel, 2020-03-05:
Excellent suggestion. I had to configure a AMPQ server and set different Exchanges for different publishers in order to have a kind of additional property for the message.
Your solution will be a much better way to do it.
Thank you

alexisfouqueteuroplacer, 2020-03-17:
Hi Tim,

I totally agree with your analysis. I have made some tests by using the "routing key" property. I assume "application properties" and "routing key" are the same thing but I am not an AMQP expert.
Can you confirm?.

Anyway my test was to automatically set the Topic of the CFX message as the routing key and it worked, so I can say that I validate the feasibility.

I will discuss this topic at the next A-Team meeting (19th March).

timburke, 2020-03-17:
> I have made some tests by using the "routing key" property

Hi @alexisfouqueteuroplacer, thanks for taking the time to test this and verify it.

I think this is a topic that will require a bit of discussion and background to find the best option so I'm going to use this comment to add some AMQP context for the broader audience that can hopefully help advance the discussion because the distinction between `routing key` and `target` vs `application properties` has changed between AMQP 0.9.1 and AMQP 1.0 and is a bit nuanced.

IPC-CFX mandates AMQP 1.0, however, I think most users probably use RabbitMQ as their broker (which the CFX working group recommends in their Getting Started guide).  RabbitMQ is a native 0.9.1 broker and supports AMQP 1.0 via a plugin that maps the AMQP 1.0 concepts down to AMQP 0.9.1 concepts so a lot of things that "don't make sense" generally in AMQP 1.0 do make sense if you're careful and are using RabbitMQ as your broker.

### Background - `target` vs `routing key`

#### AMQP 1.0 vs AMQP 0.9.1

It's tempting to think of AMQP 1.0 as a "minor version bump" over AMQP 0.9.1 but its actually not.  It's a fairly radical rethinking of AMQP concepts.

In particular, in AMQP 0.9.1 a message has a `routing key`, which is a string that indicates how the message should be routed to receivers.  In AMQP 0.9.1 a message is published to an `exchange` with a `routing key`.  The interpretation of the routing key is 100% determined by the exchange and you can implement a custom exchange to do whatever you want but generally people don't and one of 3 policies is applied to the `routing key`:
- `topic`: typical pub/sub style message routing where subscribers put filters on parts of the routing key that they care about and parts that they don't and then they receive all messages that match their wildcards.  This is more-or-less directly equivalent to how MQTT works.
- `direct`: receivers receive only those messages whose routing key exactly matches a fixed string that they define.
- `fanout`: receivers receive all messages.

AMQP 1.0 is different.  There are no exchanges and (as far as I can tell based on reading the standard), **no routing keys either.**  I believe this change was made to allow AMQP 1.0 to work in point-to-point contexts without a broker in the middle so it couldn't mandate the concept of an exchange as part of the standard because that's inherently a "broker" concept and AMQP 1.0 was intended to also support situations where there are no brokers.

Instead of publishing with a target `exchange` and  `routing key`, AMQP 1.0 just includes a `source` and `target` string that are opaque values whose interpretation can be whatever the source and receiver agree on.

#### IPC-CFX Use of AMQP 1.0

IPC-CFX mandates that it works on top of AMQP 1.0, which means that it does not have routing keys.  It does have `target` strings though.  However, all IPC-CFX machine vendor implementations that I have seen set `target` as follows:
- `target` is a fixed value hardcoded into the factory machine and the same for all messages (set by the machine owner during setup time and not dynamic in any way)
- `source` is a fixed value that is basically a unique identifier for the factory machine

So, while it's possible to use the `target` string to automatically set the routing key to include topic information on RabbitMQ brokers, this is not compatible with the way IPC-CFX machines currently set the target since it would require that the message target be different for each message.

#### AMQP 1.0 on RabbitMQ

I think one of the points of confusion here is due to the specific way that the RabbitMQ message broker supports AMQP 1.0 via a standard plugin.  RabbitMQ **is not** inherently an AMQP 1.0 broker.  It is an AMQP 0.9.1 broker (with exchanges and routing keys).  So, it needs to map all of the AMQP 1.0 concepts (like `target`) to AMQP 0.9.1 concepts so it can deal with them.

The way RabbitMQ chose to interpret the AMQP 1.0 concept of `target` is as `/<exchange>/<routing key>`  so, if you know you are sending an AMQP 1.0 message to a RabbitMQ broker, you can construct the target string in a such a way as to recover the `exchange` and `routing key` concepts from AMQP 0.9.1.

See [1].

### Header Exchanges

As a workaround for the fact that IPC-CFX machines already have a defined use of `target` and hence routing key that doesn't include filtering by topic or other properties, I was looking for a backwards compatible option that would allow for opting-in to complex message routing without requiring a reinterpretation of any current behavior.  

This is why I thought to use the headers instead of routing key (though @alexisfouqueteuroplacer I agree with you that both could be used to accomplish the same purpose).  

AMQP 1.0 defines key/value headers that you can use to route messages and RabbitMQ maps those to a "Headers Exchange" concept that lets you do that routing even if you use RabbitMQ as your broker implementation.

A header exchange lets you filter messages based on rules that apply to their header fields independent of what the routing key is, which is why it can be backwards compatible with the way IPC-CFX implementations currently set `target` to a fixed value and hence `routing key` also to a fixed value.

See [2].

### Options

[... cut ...]
    For example, consider a performance tracking system that just shows a simple OEE calculation.  Using IPC-CFX this could be an entirely general and automatic thing.  You subscribe to the performance messages and group them per line to show machine performance per line.  However, right now there's no reliable way to know which line a given machine belonged to without hardcoding that information in your application or pulling it from a separate database.

### References
[1] https://github.com/rabbitmq/rabbitmq-amqp1.0#routing-and-addressing
[2] https://www.rabbitmq.com/tutorials/amqp-concepts.html#exchange-headers