IPCConnectedFactoryExchange/CFX #57: What is the purpose of myHandle? how can i create an exchange/queue and receive the message published??

Abhishek150995, 2020-02-28:
string myHandle = "MyCompany.MyMachineModel.MyMachineSerialNumber";
        string myBroker = "amqp://cfx.aiscorp.com:5672";
        string myExchange = "/exchange/AegisCloud";
            theEndpoint = new AmqpCFXEndpoint();
            theEndpoint.Open(myHandle);
            theEndpoint.AddPublishChannel(new Uri(myBroker), myExchange);

syrandel, 2020-03-05:
Hi,
In order to create an exchange or a queue, you need to administrate a AMPQ server.
If you are developping software for a publishing endpoint, for example, a machine wishing to communicate its status, you publish to a AMQP channel, using an exchange which was supplied to you, the AegisCloud one for example.
If you are programming to exploit the endpoints data, for example to receive a machine status, you need to subscribe to an existing AMQP server channel queue. The Aegis server queues are protected and are for Aegis own use.
If you wish to have control over the whole process, you have to create your own AMQP broker. With RabbitMq, it is not very difficult.
I hope I understood correctly your question.

simon-smith, 2020-03-05:
You might find this document useful.
https://cfx.ipc.org/files/IPC-CFX-AMQP-Guide-v1_0.pdf

Abhishek150995, 2020-03-07:
> Hi,
> In order to create an exchange or a queue, you need to administrate a AMPQ server.
> If you are developping software for a publishing endpoint, for example, a machine wishing to communicate its [...]

Thanks for your valuable reply. But I am concerned to create an exchange/queue on my local RabbitMQ server programmatically.
Does the CFX-SDK library has the functionality to create Exchange/Queue and bind them without the use of the Web Console of RabbitMQ.

Abhishek150995, 2020-03-07:
> You might find this document useful.
> https://cfx.ipc.org/files/IPC-CFX-AMQP-Guide-v1_0.pdf

Thanks for your valuable reply. I read the document provided by you.
But I am concerned to create an exchange/queue on my local RabbitMQ server programmatically.
Does the CFX-SDK library has the functionality to create Exchange/Queue and bind them without the use of the Web Console of RabbitMQ.

DShiles, 2021-02-02:
Curious if you discovered a way to bind Exchange/Queue programmatically with the CFX SDK? 

I also read the IPC-CFX-AMQP-Guide-v1_0, but even though the documentation on the last page says "To ensure correct routing it is good practice to bind the queue to an exchange. It is possible to do this with the CFX SDK code", I didn't see any methods in the SDK that would allow this.