IPCConnectedFactoryExchange/CFX #83: CFX for LabVIEW

cmalpure, 2020-06-01:
Hi,

I am new to CFX standard and still trying to get a handle on the basics. I read through the CFX Demo and installed the SDK through NuGet Package available on Github. If I understand correctly the CFX SDK (a .Net library) is available that can be called in other software platforms. I am primarily a LabVIEW developer and want to implement CFX messaging through LabVIEW.

So far I have been trying to get the CFX Demo working from LabVIEW but I have not been able to connect to the host site. Maybe I am calling the wrong Properties/Methods. I was wondering if anyone here has used/called the SDK from LabVIEW and successful in getting the Demo to work. If yes, I would appreciate if you can share your LabVIEW code OR give suggestions to help me get through the issue.

Thanks,
Chetan

engrx2019, 2020-06-01:
> This Message originated outside your organization.

Hi Chetan,

I created a dll in visual basic which deals with CFX. Than I used the dll 
in labview.

Regards
Imran Ahmad
Sr. Electrical\Software Engineer
Specialty Coating Systems
317-244-1200 ext. 0292

cmalpure, 2020-06-01:
Hi Imran,

Thank you for your response. I was under the assumption that one could use the CFX dll (that takes care of the CFX) from the SDK and use it in LabVIEW. The dll you created, is it different than the dll that comes with the NuGet Package?

~Chetan

engrx2019, 2020-06-01:
Hi Chetan,

I think you should be able to call functions directly from Labview, but I 
did not try it.. It was easier for me to creating a dll.

Regards
Imran Ahmad
Sr. Electrical\Software Engineer
Specialty Coating Systems
317-244-1200 ext. 0292

<comment@noreply.github.com>
Date:   06/01/2020 01:24 PM
(#83)

cmalpure, 2020-06-03:
Hi,

I have followed the instructions given in the below link to install cfx sdk.
[http://www.connectedfactoryexchange.com/CFXDemo/sdk/html/cfb57aac-b696-4c6c-b94b-f034b37daf26.htm](url)

As given in the Getting Started page in the above link, I intended to publish a channel to a CFX broker and send a message to the Demo page by following the steps. The only difference is that I am calling the SDK (.Net library and its objects) from LabVIEW instead of using C# in Visual Studio. However, I have been unsuccessful in achieving the task and I keep getting a "NullReferenceException" error at the _AddPublishChannel_ method. 

I have gone through the Reference guide to understand the CFX functions and tasks as best as I could. As I mentioned I am new to the CFX standard. I have attached a screenshot of my LabVIEW code that calls the SDK functions through .Net constructor node. 

I would greatly appreciate if anyone can point me to the right direction and let me know what I am missing.

Thanks,
Chetan

[image]

cmalpure, 2020-06-09:
I am still getting a Null Reference Exception Error at the AddPublishChannel method. In regards to that a follow up question:

In the _AmqpCFXEndpoint_ class, I noticed there are two _AddPublishChannel_ methods: 
1. AddPublishChannel(AmqpChannelAddress, String, X509Certificate)
2. AddPublishChannel(Uri, String, String, X509Certificate)

Which one of the above two methods are called in the example code given in the "Getting Started with SDK page"? Here is the link to that page file:///C:/Projects/CFX-master/CFX-master/docs/html/cfb57aac-b696-4c6c-b94b-f034b37daf26.htm

What value should go in the argument "String" of each of the above methods?

In LabVIEW, here are the two AddPublishChannel methods as shown in screenshot:
[image]

I look forward to responses.

Thanks,
Chetan

HaythemLtifi, 2021-06-17:
Hello , i am too trying to set cfx in labview and i am stuck , can i know what your results are? did you manage to pull it off?
Thank you

1179597516, 2024-10-08:
Hello , i am too trying to set cfx in labview and i am stuck , can i know what your results are? did you manage to pull it off?
Thank you

1179597516, 2024-10-08:
Hello , i am too trying to set cfx in labview and i am stuck , can i know what your results are?Thank you