IPCConnectedFactoryExchange/CFX #99: PCBSelectiveSoldered to simple for our machines

xFirestorm, 2020-11-09:
Hi.
at the moment i am implementing PCBSelectiveSoldered in our project. 

The problem is that the model is too simple for our machines.
First of all let me explain how our machines look like.

They can have multiple soldering stages, preheating stages and fluxing stages.
A soldering stage may have multiple axis systems. On an axis system there can be one or more soldering heads. 
A flux stage may also have multiple axis systems. Each of them can have multiple bottles with different pressure and multiple flux heads with different consumption.

### ZoneData not complex enough
The PCBSelectiveSoldered class has the property SolderedPCBs which is a list of SelectiveSolderedPCB. The SelectiveSolderedPCB class has the property ZoneData which is a list of ZoneData. The ZoneData look like [this](http://www.connectedfactoryexchange.com/CFXDemo/sdk/html/T_CFX_Structures_SolderApplication_ZoneData.htm).

So if I get everything right, the idea is that for every stage the pcb went through, you create a ZoneData object and fill it with the data of the stage. 

Some examples to explain the problem in my mind:
- Let's assume a pcb went through a soldering stage. In this stage, there were two soldering heads involved. That means there are 2 Bath_Temp values, but i can only set one. I could split that into two stages, but that somehow doesn't feel right, since it was actually just one stage.
- There is a similar problem with the preheating stages. A preheater in our machine contains multiple segments. These segments have their own temperature and heating time (which can be different from segment to segment of one heater in the process). Again, I could split that into one stage per segment, but that would represent reality again because these segments don't heat the pcb after each other. Instead, they do it at the same time.

So what is the best solution for this? Should I split the phyiscal stages into seperate stages in the model? Should I just send the temperature of one bath, and ignore the values of the second bath? 

For me, the best solution would be to adapt the models to allow more complex data. For example we could either make values like the Bath_Temp a list, or instead we could add a deeper level for data with some kind of SubStageSequence property. I would offer to implement it and create a PR. But first, i wanted to discuss the architecture.

### HeaderData: Propertys that belong to stages
The PCBSelectiveSoldered class has the property HeaderData which is of type SelectiveSolderData. The SelectiveSolderData looks like [this](http://www.connectedfactoryexchange.com/CFXDemo/sdk/html/T_CFX_Structures_SolderApplication_SelectiveSolderData.htm).

So when I understood that correctly, the idea is that for a soldered pcb, there is one single value of nitrogen flow (Nitrogen_Flow). 
Again, in our machines, there are multiple of these values.

### Naming of propertys
This is not a real problem but something strange. Why do the property names have an underscore in their name? Is there a reason for that?

### We would offer to implement the changes if you don't have enough time for that right now

xFirestorm, 2020-11-12:
In the following comments i will add stuff that i find while implementing it somehow in our project.

ZoneData has the propertys Fid_XCorrection and Fid_YCorrection. We also have an angle correction. So i would suggest to add Fid_AngleCorrection.

MichaelFordUK, 2020-11-12:
Hi - It is great that you are implementing CFX - welcome to the team!  I have read through your comments, and the CFX A-team has also had a discussion. Your suggested changes look good, and potentially can be included for the next release of CFX, which is v1.3. For this version, all content should be finalized by the end of the this month. We would like to invite you to create a pull request with the ideas that you have, so that the A-team can review them and discuss them on our next call. It may be an idea for you to join that call as well if there are any questions.  Please bear in mind that we are trying to avoid making changes to CFX that would "break" existing implementations. Creating additions that add value are no problem. I am not sure exactly who you are, so I would encourage you to use IPC Works for communication. If you are not yet signed up on there, you can sign up for free, and join the IPC CFX (2591) committee. You can also contact me directly (as I am not often in Github) at mford@aiscorp.com. I am currently chairing the CFX A-team activity.  Many thanks for your ideas and contribution. Let me know if you have any questions.  Michael Ford.