IPCConnectedFactoryExchange/CFX #106: WorkCompleted async response

reget-k, 2020-12-09:
Hi,
Currently, CFX WorkCompleted message has no response message described in CFX standard.
In our use case, when AOI sends "WorkCompleted" to MES interface, AOI operator should see feedback from MES. MES Response will contain information, how message processing ended in MES interface (Ok/Error)
What could be best CFX message to use for this purpose?

Siox0911, 2021-03-23:
After you send UnitsInspected or WorkCompleted and you will further a response from a MES. You can use ValidateUnits RR from the CFX.InformationSystem.UnitValidation Namespace to the MES. The MES will response.

simon-smith, 2021-04-06:
Hi @reget-k 
Does this answer from @Siox0911 resolve your question/issue? If not can you please provide us with a more detailed requirement and the CFX standards team will have a look for you.
Simon

katonori2501, 2021-09-28:
I am also interested in this topic.
My customer requests us to "overwrite" the result (OK / Error).
If we can use ValidateUnitsRR for this, I'd like to revise the description of this message like this.

Description
Process endpoints, where configured, should send this request before allowing the next unit to **enter or exit** the process.