IPCConnectedFactoryExchange/CFX #113: Abilty of CFX to transfer unique board info like MAC address, device firmware, etc.

CharlesARobinson, 2021-02-08:
I am an applications engineer not a developer so please bear with me. 

On our test equipment we currently are able to create custom routines that, after successful validation of a UUT serial number, can download from the customer's MES system unique items like a MAC address or firmware to be programmed into the UUT at test. The validation and download of anything unique is completely custom, based on the customer's MES implementation and their needs. 

So I'm wondering when a customer moves to an IPC-CFX implementation, does CFX allow for transmission of this kind of data, either as part of a successful validation or as a separate action taken after a successful validation. As I imagine that CFX would be the sole messaging mechanism with the MES system, I do not envision using a custom app outside of CFX doing this. 

I just wonder if the standard allows for this somehow.

johnjwalls, 2021-03-23:
Hi Charles.  Yes, CFX does support this.  Check out the [UnitsPersonalized ](http://www.connectedfactoryexchange.com/cfxdemo/sdk/html/T_CFX_Production_Assembly_UnitsPersonalized.htm) message in the CFX SDK reference guide.

CharlesARobinson, 2021-04-14:
Hi John, Thanks for your reply. I am interested in a transfer of data from the Factory to the Endpoint. UnitsPersonalized covers transfer from the Endpoint to the Factory, so in the opposite direction than I'm looking for.

I've raised this with the committee. It will be discussed and added in 1.4