IPCConnectedFactoryExchange/CFX #117: Amqp.AmqpException: 'The address field cannot be empty'

micha-1987, 2021-04-09:
Dear folks,

I am quite desperate already.
I am trying to execute a request to a second CFX application.
This application ran on the same PC and I also tried it on another PC.

I am always getting the error: Amqp.AmqpException: 'The address field cannot be empty' when executing the request.
I mainly used the code being provided by the CFX SDK onboarding guide.

`
string myHandle = "MichaelsCompany.VirtualCOntrolSystem.9876543210";
requestEndpoint = new AmqpCFXEndpoint();
requestEndpoint.Open(myHandle);
string targetEndpointHostname = "127.0.0.1:5672"; 
                                                        
string remoteUri = string.Format("amqp://{0}", targetEndpointHostname);
CFXEnvelope request = CFXEnvelope.FromCFXMessage(new GetEndpointInformationRequest()
            {
                CFXHandle = "MichaelsCompany.VirtualSMTMachine.0123456789";
            }) ;

AmqpCFXEndpoint.RequestTimeout = TimeSpan.FromSeconds(20);
CFXEnvelope response = requestEndpoint.ExecuteRequest(remoteUri, request); // --> Here is where the exception is raised
`
I already checked if the second application is listening on the given address and indeed it does,
Even stranger is it when I put in a another existing IP as hostname. Then I am getting a timeout and not the AMQP exception.
This annoys me. No request is recognized at the second application (Endpoint_OnRequestReceived). So I assume it is not sent correctly but I am not pretty sure.

Hopefully somebody can help me.

engrx2019, 2021-04-13:
I think request is not transmitted if there is an error. I was getting error  {"Attach rejected: {address_not_utf8_string,undefined}"} | Amqp.AmqpException until I added source and target handles. See if this fixes your issue.

request.Target = "MichaelsCompany.VirtualSMTMachine.0123456789"
request.Source = myHandle

Now I am getting following error even though the queue is set as Durable.

 {"PRECONDITION_FAILED - inequivalent arg 'durable' for queue 'targetHandle' in vhost '/': received 'false' but current is 'true'"} | Amqp.AmqpException

micha-1987, 2021-04-20:
Hello engrx2019,

yes, indeed it fixed my issue.
Many thanks!
For me it works together with rabbitmq but the queue must also not be of type quorum.
Obviously only classic queues having the durable option works.
But actually I dont know why this restriction exists. Why not sending messages to a temporary queue?

Anyhow, thanks again :)