IPCConnectedFactoryExchange/CFX #125: IPC-CFX SDK is incomplete in regards to examples for messages like UnitsArrived, ValidateUnitsRequest, etc.

CharlesARobinson, 2021-05-02:
If I look at the messages UnitsArrived, UnitsDeparted, ValidateUnitsRequest, ValidateUnitsResponse there are properties listed that are not illustrated in the example messages. As such one cannot know the structure of the messages when these properties are included. There could be other messages affected by this omission as well. 

The properties are HermesIdentifier, Lane and PrimaryIdentifier. 

See http://www.connectedfactoryexchange.com/CFXDemo/sdk/html/T_CFX_Production_UnitsArrived.htm as an example. Only two of the five properties are used in the example.

Siox0911, 2021-05-06:
This is wrong. The Structur is the CFXEnvelope or the CFXMessage himself. So you can check every MessageBody with
```C#
if(cfxEnvelope.MessageBody is UnitsArrived unitsArrived)
{
    //your code for UnitsArrived
   unitsArrived.HereIsTheProperty //This could be null, but is it definitely default
   if(unitsArrived.HereIsTheProperty == default)
   {
      //It's the default value
   }
}
```

CharlesARobinson, 2021-05-06:
I'm sorry but I don't understand your response. I am referring to the way that CFX messages are documented in the SDK and noticed that these particular messages seem to me missing something in their documentation. If there is some unique way that these messages behave that explains the way that they are documented then I am not clear on it.

Siox0911, 2021-05-06:
The CFX is nothing for people, if they understand only the theory of programming. The documentation is just an example, not the reality.

Siox0911, 2021-05-06:
Everything is possible, nothing is neccesary. You can determine everything with the partners. And if you have good programmers, then you can configure that per endpoint without reprogramming.