IPCConnectedFactoryExchange/CFX #131: Broadcast Request handling

cgs-jcdubois, 2021-06-16:
Hi,

Does anyone have an example on how the broadcast request is currently working?  So let's say a WhoIsThere request is sent.  Each endpoint would receive the WhoIsThereRequest through an exchange.  Where is it expected to reply with the WhoIsThereReponse, what path should it follow?  Do we have all the required information provided in the Message properties?  Would it be in the Handle information?  In the reply-to?  Somewhere else?

Thanks,
Julie

alexisfouqueteuroplacer, 2021-07-28:
Hi Julie,

As far as I know, there is no information in the broadcast requests to indicate where to reply. It is up to the endpoint to know on which Exchange to answer back. So it is more a upper level configuration concern.

Best regards

Alexis

madi-cogi, 2021-09-08:
Hi Alexis,

Just to be sure of the expected behavior here. 
An endpoint is subscribed to zero or more channels (queues in RabbitMQ) and has zero or more publish channels (exchanges in RabbitMQ).
If any one of the subscribed channel receives a WhoIsThereRequest the endpoint will proceed to publish a WhoIsThereResponse to all its configured publish channels.

The original sender of the request will monitor for WhoIsThereResponse with the same Request/Correlation id?

Is this accurate?

RobertHudjakov, 2022-05-31:
> If any one of the subscribed channel receives a WhoIsThereRequest the endpoint will proceed to publish a WhoIsThereResponse to all its configured publish channels.

Seems accurate. There's a hint in chapter 5.1.3 (CFX Message Types). Broadcast response is from Destination(s) -> Source(s).