IPCConnectedFactoryExchange/CFX #132: Avoid queues auto creation

alexisfouqueteuroplacer, 2021-06-18:
Hi every one,

When AddSubscribeChannel is called, with a queue that doesn't exist in the RabbitMQ broker, it is automatically created. Is there a way to avoid this behaviour and just send an exception instead ?

Thanks!

Alexis

flex-altmraut, 2021-06-22:
Hi Alexis,

I would say the best way is to control it by the RabbitMQ user permissions. There you can define rules with regular expressions which queues/exchanges are allowed to create and access. This is how we control naming conventions and auto creation.

Simple example:
[image]

This would only allow to create and access queues which are starting with "cfx", e.g. "cfx-machine1" and would not allow to auto create a queue called "machine1"

BR
Michael

alexisfouqueteuroplacer, 2021-07-02:
Hi Michael,

Thanks for your answer, this naming convention control is very interesting! But it doesn't resolve an issue we have with this scenario :
- We have a configuration software, with which we can define the endpoints and the queues names, among other parameters. This software is also able to send commands to RabbitMQ to create/remove the queues. For example we create the endpoint ENDPOINT_001 and the queue QUEUE_001.
- When the equipment runs, it reads the configuration and connect to the queue (QUEUE_001)
- If we change the configuration, for example we rename the queue from QUEUE_001 to QUEUE_002, the configuration software will delete the old queue, and create the new one.
- The problem is that as the equipment is still running, it will automatically create the QUEUE_001 ! We would like to just make the equipment disconnect unless it is started again (where it will connect to QUEUE_002 this time).

Do you or anyone has a solution for this?

Best regards,

Alexis

flex-altmraut, 2021-07-12:
Hi Alexis,

as far as I know there is currently no other way beside the user permissions via RabbitMQ.
In your case you could create a user for each machine and give permissions to the exact queue/exchange name.

**Example:**

User: machine001
Read/Write permissions: QUEUE_001

If you want to change the queue name for this machine you would remove permissions to QUEUE_001 first, delete QUEUE_001, add QUEUE_002 and grant permissions to QUEUE_002 to machine001.

BR
Michael

alexisfouqueteuroplacer, 2021-07-28:
Hi Michael

Thanks Michael, that's a pretty clever work around!

Best regards,

Alexis