IPCConnectedFactoryExchange/CFX #143: Request/Response using RabbitMQ- Event not firing

quesGit, 2021-09-28:
Hi all,

Using amqpendpoint class and rabbitmq to send and consume messages. When I send a message in the way of amqpendpoint, i can able to send the messages but the problem is "TheEndpoint_OnRequestReceived" event not firing automatically when there is a CFX message. What is the reason?How to solve it?
```
void Send(){
 if (!IsOpen) return;
            CFXEnvelope request = CFXEnvelope.FromCFXMessage(new AreYouThereRequest()
            {
                CFXHandle = MyCFXHandle,

            });
              MyEndpoint.Publish(request);
             AddToResults($"MESSAGE PUBLISHED:\n {request.ToJson(true)}");
}
```
```
void receive(){
if (!IsOpen) return;

 MyEndpoint.OnRequestReceived += TheEndpoint_OnRequestReceived;
}
```
 ```
private CFXEnvelope TheEndpoint_OnRequestReceived(CFXEnvelope request)
        {
                  if (request.MessageBody is AreYouThereRequest)
                {
                    CFXEnvelope response = CFXEnvelope.FromCFXMessage(new WhoIsThereResponse()
                    {
                        CFXHandle = MyCFXHandle,
                        RequestNetworkUri = CFXDemoEventBroker.ToString(),
                        RequestTargetAddress = TargetMESEndpointUri
                    });
                    response.Source = MyCFXHandle;
                    response.Target = request.Source;
                    return result;
                    AddToResults($"MESSAGE PUBLISHED:\n {response.ToJson(true)}");
                }
                     return null;
        }

```

henryclon, 2022-01-11:
I too have this question, if you found out how it works please let me know!

SarumathiGetech, 2022-01-21:
Hi

Broadcast request/repsonse work from endpoint to endpoint with the help of broker. They are called requests but they are actually events requiring a response... i.e : AreyouThereRequest/Response. 

On the sending side you need to add a publish channel using endpoint.AddPublishChannel() with the correct uri to reach the broker .i.e Publish Address (exchange name).

On the receiving side, you are suppossed to used OnCFXMessageReceived to receive the AreyouThereRequest message. Then this received can send the response back with endpoint.Publish the AreYouThereResponse message .