IPCConnectedFactoryExchange/CFX #148: Close an endpoint inside a callback call

alexisfouqueteuroplacer, 2021-12-20:
Hi,

Here's the situation : An application treats the received CFX messages in a callback (let say `my_OnCFXMessageReceived`). But at a moment, an error occurs in the callback, and we need to stop listening and let the next messages stacking in the AMQP queue. In other words, we want this callback call to be the last one before we re-open the endpoint.

Calling the `Close` method inside the `my_OnCFXMessageReceived` callback doesn't work because `Close` waits for `my_OnCFXMessageReceived` to end... Throwing an exception doesn't work either because it is caught without doing a Close.

Is there any solution to this kind of scenario?

Thanks,

Alexis

Siox0911, 2022-02-16:
If you want to close an endpoint which is running in a other context (thread). You can't do this from an other thread directly. This is thread unsafe. But you can delegate this to the UI thread. Example is for WPF.

```C#
private void Control_OnCFXMessageReceived(CFX.Transport.AmqpChannelAddress source, CFX.CFXEnvelope message)
{
    Application.Current.Dispatcher.Invoke(delegate
    {
        //Close the endpoint here
        Endpoint.Close();
    });
}
```

alexisfouqueteuroplacer, 2022-02-16:
Hi Siox0911. Thanks for your answer.

I understand your point, but in your example we still have the same problem, the Close() function will wait the callback to end so we are stuck... (I believe that Application.Current.Dispatcher is a synchronous call).

Siox0911, 2022-07-25:
This has nothing to to with CFX. Close.

Siox0911, 2022-07-25:
> I believe that Application.Current.Dispatcher is a synchronous call
The Invoke call just puts this on the thread responsible for the caller.

alexisfouqueteuroplacer, 2022-07-25:
Anyway in our real use case we don't have any UI, it runs as a service... But I think we should think without any specific platform or scenario. The lack I am pointing is the ability for the reception callback to request a close and avoid any new message consumption from the amqp queue.