IPCConnectedFactoryExchange/CFX #168: How do I send and receive user-defined information?

jeungyeop10, 2022-07-12:
How do I send and receive user-defined information?

Can you send and receive information other than the defined message?

For example, equipment A transmits the information "abcd" to the customer's system, What should the customer's system do when it tries to transmit the information "1234" to Equipment A?

Siox0911, 2022-07-25:
In principle, you can abuse CFX for this. Nevertheless, one should try to stick to the standard. Why? Well, if an image on the Internet is an audio file, then it's no longer an image. Not every application will then understand the image and many will even interpret it as an error. This does not lead to the goal. This is how CFX created a standard. So the goal is not to circumvent the standard, but to apply the machines and systems to it. This requires a little rethinking, but leads to the goal.

Siox0911, 2022-07-25:
For example you can abuse Recipe for everything.

```C#
            var recipeRequest = new CFX.Production.GetRecipeRequest
            {
                RecipeName = "abcd"
            };

            var recipeResponse = new CFX.Production.GetRecipeResponse
            {
                Result = new CFX.Structures.RequestResult { Result = CFX.Structures.StatusResult.Success },
                Recipe = new CFX.Structures.Recipe { RecipeData = Encoding.ASCII.GetBytes("1234") }
            };

```