IPCConnectedFactoryExchange/CFX #171: Connecting to an Azure Service Bus Certificate null Error

MercyPillow, 2022-08-09:
Im trying to send messages to an Azure service bus that i can connect to and communicate with using AMQPNetLite using the regular OAuth Method "amqps://{policy}:{SASKey}@{nameSpace}.servicebus.windows.net" on port 443 (forwarded) however when using the CFX dll to do this im getting "One or more errors occurred" 

After breaking down the errors and replacing functions from the CFX.dll i found im getting the probem here: 
https://github.com/IPCConnectedFactoryExchange/CFX/blob/f3435a7fa164d1374e1379083baccd239c4d1887/CFX/Transport/AmqpCFXEndpoint.cs#L532

"fact.SSL.RemoteCertificateValidationCallback = ValidateServerCertificate;"
Validating the certificate when using AMQPS fails because im not passing a certificate, its validated with SAS policyName:SASKey

removing the Certificate validation stops the issue and connects just fine and returns True, but this is in the AddSubscribeChannel method too whichs prevents a connection being made to listen for messages.

tested AMQPNetLite code sending and recieving to the service bus without a X509Certificate, after looking through the CFX repo its the same just not barebones (amqps is required or the connection is thrown out) 
```
            string PolicyName = "cfxPolicy";
            string SASKey = "someSASKey";
            string nameSpace = "{nameSpace}.servicebus.windows.net";
            string connectionString = $"amqps://{PolicyName}:{SASKey}@{nameSpace}/";

            Address address = new Address(connectionString);
            Connection connection = new Connection(address);
            Session session = new Session(connection);

            Amqp.Message sendMessage = new Amqp.Message("Hello AMQP!");
            SenderLink senderlink = new SenderLink(session, "sender-link", "cfx");
            senderlink.Send(sendMessage);
            Console.WriteLine("Sent Hello AMQP!");

            senderlink.Close();

            Source source = new Source()
            {
                Address = "cfx",
                Durable = 2
            };
            ReceiverLink receiver = new ReceiverLink(session, "cfx", source, null);

            Console.WriteLine("Receiver connected to broker.");
            Amqp.Message message = receiver.Receive();
            Console.WriteLine("Received " + message?.GetBody<string>());
            receiver.Accept(message);

            receiver.Close();
            session.Close();
            connection.Close();
```

how does one get around this or are we forced to use a certificate unnecessarly along side other credentials?

MercyPillow, 2022-08-10:
im blind, missed disabling certificate validation
```
Endpoint.ValidateCertificates = false;
```

Error message of "One or more errors occurred." could be imporved however or a null certificate check added.

closed 2022-08-10 by MercyPillow

IsabellaLicht, 2023-08-30:
Hi @MercyPillow , I want to use Azure Service Bus as a broker too. If I use AMQP everything works fine... But with the  CFX.Transport I have the following problem: I can connect to the Azure Service Bus (Successfull Request in Azure and TestSubscribeChannel /TestPublishChannel is true) but I'm not able to see new published messages in Azure. Can you help me? is "myExchange" in the right order?
Here is my code:

```
        class Endpoint
        {
            AmqpCFXEndpoint thisEndpoint;
            string myHandle = "100.TE.61000";
            string myBroker = $"amqps://name:key@host.servicebus.windows.net/";
            string myExchange = $"/topic/subscription";

            public void Openwithendpoint()
            {
                thisEndpoint = new AmqpCFXEndpoint();
                thisEndpoint.Open(myHandle);
                Console.WriteLine(thisEndpoint.IsOpen);
                thisEndpoint.ValidateCertificates = false;
            }
            public void Publish()
            {
                thisEndpoint.AddPublishChannel(new Uri(myBroker), myExchange, "bmkprod.servicebus.windows.net");
                List<CFXEnvelope> messages = new List<CFXEnvelope>();
                CFXEnvelope env = new CFXEnvelope(new CFX.ResourcePerformance.LogEntryRecorded()
                {
                    Importance = CFX.Structures.LogImportance.Debug,
                    Message = "Debug Log Entry"
                });
                messages.Add(env);
                
                bool testergebnis=thisEndpoint.TestPublishChannel(new Uri(myBroker), myExchange, out Exception error, "bmkprod.servicebus.windows.net");
                thisEndpoint.Publish(env);
               
            }
        }
```