IPCConnectedFactoryExchange/CFX #222: Include documentation and source link in the .NET core and standard packages

megla-tlanghorst, 2025-01-08:
The [.NET (Core)](https://nuget.info/packages/CFX.CFXSDK.NetCore/1.7.3) and .NET Standard packages do not include the XML comments file:
[image]

The [.NET Framework Package](https://nuget.info/packages/CFX.CFXSDK/1.7.3) does include it:
[image]

The required source code commit for source link is also missing in the metadata of all packages, so source link is unavailable:
[image]

It should look like this:
[image]

Instruction on how to set it up can be found here: https://github.com/dotnet/sourcelink/blob/main/README.md

Lastly maybe if you want to save some hassle for both yourself and your users you could also consolidate net462, netstandard1.3 and .netcoreapp3.1 (plus maybe nowadays net8.0?) into a single package like CFX.CFXSDK and MSBuild will then choose the best, compatible version automatically without any need to manually wind the right package. That looks like this then:
[image]

megla-tlanghorst, 2025-01-08:
As I understand it if you build it using .NET 8 (doesn't necessarily have be the or the only target though I think)

ChrisJorgensen, 2025-05-21:
Tim,

There has been a bug identified specific to the pull request from you that was merged into the CFX 2.0 SDK. There is a message string about this under #236. Can you review and respond to the two users on the message string?

Here is a direct link to that string: https://github.com/IPCConnectedFactoryExchange/CFX/issues/236#issuecomment-2893464717

Chris

***@***.***<https://www.ipc.org/>
***@***.***<https://www.ipc.org/facebook> ***@***.*** <https://www.ipc.org/twitter>  ***@***.*** <https://www.ipc.org/youtube>  ***@***.*** <https://www.ipc.org/linkedin>  ***@***.*** <https://www.instagram.com/ipcassociation/>
Chris Jorgensen | Senior Director, Next-Generation Standards (He/Him/His)
IPC
***@***.*** 3000 Lakeside Drive, Suite 105 N, Bannockburn, IL 60015
***@***.*** +1 847 597-2826
***@***.*** +1 773 771-9411
***@***.*** ***@***.******@***.***>
***@***.*** www.ipc.org<http://www.ipc.org/>

image.png (view on web)<https://github.com/user-attachments/assets/44184746-7f29-4aca-8489-6de13d160fe3>

image.png (view on web)<https://github.com/user-attachments/assets/89dbc96a-12bb-4118-a3c5-f05b1454f657>

image.png (view on web)<https://github.com/user-attachments/assets/de883877-5206-491c-aa70-eecf84f6a6c8>

image.png (view on web)<https://github.com/user-attachments/assets/12ac6a40-d65a-4c79-9a75-301fe10db26b>

image.png (view on web)<https://github.com/user-attachments/assets/de24f569-e19f-4464-bcc9-cbdf522f390e>