IPCConnectedFactoryExchange/CFX #227: cfx are always better than reality

huangyu371, 2025-02-18:
Dear CFX Team,

Our company is implementing and utilizing CFX to improve efficiency in our factory. However, we have encountered many issues, as stated in the subject line. The quality of the programs developed by our vendors varies significantly, and we have identified the following problems:

1. A significant discrepancy between data timestamps and real-world time.
2. Incorrect retrieval of the HermesIdentifier.
3. Missing or out-of-sequence messages.

These issues are not immediately visible to the naked eye. Given the large amount of data, it takes considerable time and effort to uncover them. In the future, we plan to add several new production lines, which means dealing with new vendors. Apart from verifying the accuracy of the data in the messages, having to conduct such extensive investigations is extremely exhausting. More importantly, we find ourselves repeatedly facing the same problems.

The process from vendor feedback to issue resolution is excessively time-consuming. While you provide the system and licenses to vendors for development, it appears that there are no guarantees regarding the quality of the software they produce. As end users, instead of benefiting from the system, we are burdened by these issues.

I would like to ask whether CFX provides any standards or regulations to ensure the quality of vendor-developed programs. This would help ensure data integrity and software stability for end users.

I apologize for the complaints, but there are simply too many issues. After sending this email, I will have to return to addressing these problems. Even if fixes are made, they will only come in the next version.

Best regards,

ChrisJorgensen, 2025-02-18:
We would like to help you with these issues. Please email me directly to ChrisJorgensen@ipc.org.

ChrisJorgensen, 2025-02-19:
> If I reply to the email, will I receive a perfunctory response or a concrete answer? I just want to know whether your company provides any guarantees regarding data and software quality to ensure user rights are protected. Otherwise, it feels like we're simply buying a bunch of bugs from the vendor and going on a treasure hunt.
>
>
> If I reply to the email, will I receive a perfunctory response or a concrete answer? I just want to know whether your company provides any guarantees regarding data and software quality to ensure user rights are protected. Otherwise, it feels like we're simply buying a bunch of bugs from the vendor and going on a treasure hunt.

IPC does not provide guarantees regarding vendor software quality, how vendors are populating data or user rights protections. These would be agreements between you and the vendors.

IPC offers the IPC-CFX-2591 Qualified Products List (QPL), where vendors can have equipment listed once it has passed a third-party qualification audit to show the CFX implementation is in accordance with the requirements of IPC-2591 for syntax, order of messages, requires mandatory message for specific equipment and AMQP for encryption.

My full contact information is below, in case you would like to reach out to me directly to further discuss your issues.

Chris

***@***.***<https://www.ipc.org/>
***@***.***<https://www.ipc.org/facebook> ***@***.*** <https://www.ipc.org/twitter>  ***@***.*** <https://www.ipc.org/youtube>  ***@***.*** <https://www.ipc.org/linkedin>  ***@***.*** <https://www.instagram.com/ipcassociation/>
Chris Jorgensen | Director, Technology Transfer (He/Him/His)
IPC
***@***.*** 3000 Lakeside Drive, Suite 105 N, Bannockburn, IL 60015
***@***.*** +1 847 597-2826
***@***.*** +1 773 771-9411
***@***.*** ***@***.******@***.***>
***@***.*** www.ipc.org<http://www.ipc.org/>

huangyu371, 2025-02-21:
Could you escalate this request? This issue has a significant impact on us as users. Without quality assurance in the program, the integrity and accuracy of the data cannot be guaranteed. This puts immense pressure on the entire CFX ecosystem, as both users and vendors will be stuck in an endless cycle of bug fixing. In such a scenario, the only true winner would be IPC-CFX, not everyone involved.

Users want to benefit from the CFX architecture and build various extended applications on top of it without worrying about data integrity and accuracy. Vendors, on the other hand, can secure more orders thanks to the quality assurance provided by CFX.

huangyu371, 2025-02-24:
Could you bring up this request to the higher-ups? This issue has a significant impact on us as users. Without quality assurance in the program, the data produced cannot be guaranteed to be clean and accurate. This puts immense pressure on the entire CFX ecosystem, trapping both users and vendors in an endless cycle of bug-fixing. In such a scenario, the only winner would be IPC-CFX, rather than everyone involved.

Users also want to benefit from the CFX framework and build various extended applications based on it without worrying about data integrity and accuracy. Meanwhile, vendors can secure more orders thanks to the quality assurance provided by CFX.

ChrisJorgensen, 2025-02-24:
We appreciate these concerns and will do our best to address them. Is the equipment in question on the CFX QPL?

Can you email me a list of model numbers and software versions for the equipment?

Chris

***@***.***<https://www.ipc.org/>
***@***.***<https://www.ipc.org/facebook> ***@***.*** <https://www.ipc.org/twitter>  ***@***.*** <https://www.ipc.org/youtube>  ***@***.*** <https://www.ipc.org/linkedin>  ***@***.*** <https://www.instagram.com/ipcassociation/>
Chris Jorgensen | Director, Technology Transfer (He/Him/His)
IPC
***@***.*** 3000 Lakeside Drive, Suite 105 N, Bannockburn, IL 60015
***@***.*** +1 847 597-2826
***@***.*** +1 773 771-9411
***@***.*** ***@***.******@***.***>
***@***.*** www.ipc.org<http://www.ipc.org/>