IPCConnectedFactoryExchange/CFX #233: Extended messages not arriving when not known.

LucVerstreken, 2025-04-09:
When I extend a CFX message and it arrives in an application which does not know this extended message, it will never arrive. It would be good that if this extended message is not known, it will deserialize the one of base classes which is a standard CFX message.

Example I have extended the ValidateUnitsRequest in the class ValidateUnitsRequestExt

In Json, you can find this type:

"$type": "MyNamespace.ValidateUnitsRequestExt, MyAssembly"

The Json deserializer will fail deserializing this string because the ValidateUnitsRequestExt is not know in this application.

Solution:

Add base class info in the message:

"$type": "MyNamespace.ValidateUnitsRequestExt, MyAssembly",
  "$baseTypes": [
    "CFX.InformationSystem.UnitValidation.ValidateUnitsRequest, CFX"
  ],
...

$baseTypes is a custom property which has to be added.
If the deserialization fails, it can check if one of the base classes is a standard CFX message, and if so serialize it to the base class.

This should solve a lot of our problems. We have to keep sender and receiver in sync according message types sent. But this is not always easy.

LucVerstreken, 2025-04-28:
Is it normal that nobody is reacting to these kind of issues? This issue is 3 weeks old.

alexisfouqueteuroplacer, 2025-05-04:
Hi, I think this @johnjwalls would be the best person to answer this question, but in my opinion, extending CFX messages is not desirable. I don't think that's the purpose of a standard because then it's no longer a standard.
If something is missing from certain messages, feel free to report it to the IPC group. If the changes are deemed relevent and accepted, they may be incorporated in a future version of CFX.
Alexis