IPCConnectedFactoryExchange/CFX #235: Direct CFX requests no longer work since CFX 2.0

alexisfouqueteuroplacer, 2025-05-05:
Since CFX 2.0, direct requests no longer work. The requester waits until the timeout (by default, 1 minute), but nothing happens on the receiver side.

After some testing, I discovered that the problem comes from the AMQP.NET library, which was upgraded from version 2.4.5 to 2.4.11 between CFX 1.7 and CFX 2.0. With version 2.4.6, the requests still works, but with versions 2.4.7 and later (i.e., 2.4.11), they don't work anymore.

It's very easy to reproduce with CFXExampleEndpoint:
[image]

johnjwalls, 2025-05-22:
Thasnk for the fix Alexis.  I merged your PR into master, and cherry picked it into the 2.0 release branch.  I just published a new nuget package (2.0.4) that references the older version.  Will take a closer look at why the newer package doesn't seem to work when I have more time.  Maybe MS will release a new package that solves the issue in the meantime :)  John

closed 2025-05-22 by johnjwalls

X-RabasF, 2025-06-06:
HI @johnjwalls ,
downgrading is not really solving the issue here, because with that this topic is reappearing:
https://github.com/IPCConnectedFactoryExchange/CFX/issues/221