IPCConnectedFactoryExchange/CFX #236: Compatibility Issues with CFX SDK 2.0.3 on .NET Standard 2.0 and .NET Framework 4.8

KeMSey, 2025-05-14:
Hi there,
 
We’re encountering some compatibility issues when testing with the new IPC CFX SDK Version 2.0.3. Here’s some context on our setup:
 
We have an application that uses a plugin-based architecture, and the IPC CFX functionality is implemented within one of these plugins. We build our plugins targeting .NET Standard 2.0.

**Update**

We recently updated the package reference in this plugin from CFX.CFXSDK.NetStandard 1.7.3 to CFX.CFXSDK 2.0.3. After the upgrade, we began seeing the following error when using the IPC CFX Qualification Tool. Specifically, when trying to open the endpoint, the following exception is logged:

2025-05-12 16:31:18.002 +02:00 [ERR] An error occured while trying to open the endpoint. Error Message: Method not found: 'Void System.IO.FileStream..ctor(System.String, System.IO.FileMode, System.IO.FileAccess, System.IO.FileShare)'. Plugin.IpcCfx.Service.CfxCommunicationService.
System.MissingMethodException: Method not found: 'Void System.IO.FileStream..ctor(System.String, System.IO.FileMode, System.IO.FileAccess, System.IO.FileShare)'.
at CFX.Transport.DurableQueue.Initialize()
at CFX.Transport.DurableQueue..ctor(String name)
at CFX.Transport.AmqpSenderLink..ctor(Uri uri, String address, String sourceHandle, AmqpConnection conn)
at CFX.Transport.AmqpConnection.AddPublishChannel(String address, Boolean connectImmediately)
at CFX.Transport.AmqpCFXEndpoint.AddPublishChannel(Uri networkAddress, String address, String virtualHostName, X509Certificate certificate)
at Plugin.IpcCfx.Service.MyAmqpCfxEndpoint.Plugin.IpcCfx.Service.IMyAmqpCfxEndpoint.AddPublishChannel(Uri networkAddress, String address, String virtualHostName, X509Certificate certificate)
at Plugin.IpcCfx.Service.CfxCommunicationService.AddChannel(String amqpExchangeValue, Uri brokerUri, String amqpQueueValue) in C:\Ersa\Repos\TransformerAddon.Plugin.IpcCfx\src\Plugin.IpcCfx\Service\CfxCommunicationService.cs:line 320
at Plugin.IpcCfx.Service.CfxCommunicationService.ConfigureAndOpenAmqpEndpointAsync() in C:\Ersa\Repos\TransformerAddon.Plugin.IpcCfx\src\Plugin.IpcCfx\Service\CfxCommunicationService.cs:line 289
at Plugin.IpcCfx.Service.CfxCommunicationService.OpenEndpointAsync() in C:\Ersa\Repos\TransformerAddon.Plugin.IpcCfx\src\Plugin.IpcCfx\Service\CfxCommunicationService.cs:line 253

**Additional Observations**
 
When building the plugin against .NET Framework 4.8, the endpoint opens, but sending a request (e.g., CFX.InformationSystem.UnitValidation.ValidateUnitRequest) times out:

2025-05-13 08:21:52.492 +02:00 [ERR] IPC CFX, could not send request because: The operation send did not complete within the allocated time 60000 for object message. Plugin.IpcCfx.Service.CfxCommunicationService.
System.TimeoutException: The operation send did not complete within the allocated time 60000 for object message.
  bei CFX.Transport.AmqpCFXEndpoint.<ExecuteRequestAsync>d__142.MoveNext()
--- Ende der Stapelüberwachung vom vorhergehenden Ort, an dem die Ausnahme ausgelöst wurde ---
  bei System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
  bei System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
  bei System.Runtime.CompilerServices.TaskAwaiter`1.GetResult()
  bei Plugin.IpcCfx.Service.CfxCommunicationService.<RequestAsync>d__17.MoveNext() in C:\Ersa\Repos\TransformerAddon.Plugin.IpcCfx\src\Plugin.IpcCfx\Service\CfxCommunicationService.cs:Zeile 166.
 
This seems similar to a previously reported issue: Issue [#235](https://github.com/IPCConnectedFactoryExchange/CFX/issues/235).
 
Finally, when building the plugin against .NET 8, everything works as expected — the endpoint opens and requests are sent successfully.
 
**Question**
 
Can you help confirm whether this behavior is reproducible on your side? It appears the SDK may not be fully compatible with .NET Standard 2.0 and .NET Framework 4.8.
 
Thanks in advance for your help!

alexisfouqueteuroplacer, 2025-05-14:
Hi. I am the author of issue #235, and I confirm that our application is using .NET Framework 4.8. I didn't tried with other flavor of .NET.
Did you try to rebuild the SDK in .NET Standard 2.0 with the AMQP.NET library downgraded to 2.4.6 ?

KeMSey, 2025-05-20:
Hi, thanks for the quick response and for confirming the framework. When comparing the two exceptions above (.NET Framework 4.8 and .NET Standard 2.0) it seems to be different issues. I tried the rebuild of the SDK targeting .NET Standard 2.0 with the downgraded AMQPNetLite (2.4.6) and I could reproduce the exception mentioned earlier.

johnjwalls, 2025-05-22:
Hi.  I attempted to reproduce this problem, but have been unable to.  To test, I created a branch new .NET Standard 2.0 Class Library.  I then added CFX.CFXSDK 2.0.3 package to the library as a dependency, and added some test code to the DLL to create an AmqpCFXEndpoint, and publish a few messages.  I then created a .NET Framework 4.8 console application, and referenced the new Standard 2.0 library I had just created.  the console app calls into the 2.0 Standard class library, and invokes the test code.  And messages are getting queued and sent without issue.  I am using Visual Studio 2022 to do this.

Could you put together a small test application on your side that re-creates the issue, and share the solution/project and code with me?  Then I can try to figure out what is going on.  Thanks.

johnjwalls, 2025-05-27:
So I am still unable to reproduce the issue, even using a .NET 8 console application.  I also containerized as a Linux docker, and it ran inside that environment ok too.  Please see the test application I am using here:  https://github.com/IPCConnectedFactoryExchange/CFXIssues/tree/main/Issue_236

closed 2025-06-04 by KeMSey